LennartHennigs/SimpleFSM #1: Simple example blows up esp32 huzzah feather

snowrodeo, 2022-05-27:
Hi, I modifed the code in the mixed transitions with browser example to add a 'yellow' light for fun.  It seems to crash my Adafruit huzzah feather 100% of the time.  

Here are the errors from the serial port:
SimpleFSM - Mixed Transition (Traffic light with button for green)

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40089278  PS      : 0x00060a30  A0      : 0x800d9405  A1      : 0x3ffb2740  
A2      : 0xcdeed2f7  A3      : 0x3ffc2ce0  A4      : 0x00000001  A5      : 0x0000009d  
A6      : 0x3ffb91e4  A7      : 0xff000000  A8      : 0x00000000  A9      : 0xcdeed2f7  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x0000009e  A13     : 0x3ffb9282  
A14     : 0x0000000a  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xcdeed2f7  LBEG    : 0x400892d0  LEND    : 0x400892db  LCOUNT  : 0x00000000  

Backtrace:0x40089275:0x3ffb27400x400d9402:0x3ffb2760 0x400d9563:0x3ffb2780 0x400d380d:0x3ffb27a0 0x400d2db6:0x3ffb27c0 0x400d9df6:0x3ffb2820 

Here is the code: 
/////////////////////////////////////////////////////////////////

#include "SimpleFSM.h"
#include "Button2.h"    // https://github.com/LennartHennigs/Button2

#if defined(ARDUINO_ARCH_ESP32)
    #include <WiFi.h>
    #include <WiFiClient.h>
    #include <WebServer.h>
#elif defined(ARDUINO_ARCH_ESP8266)
    #include <ESP8266WiFi.h>
    #include <WiFiClient.h>
    #include <ESP8266WebServer.h>
#else 
    #error "you need an ESP32 or ESP8266"
#endif

/////////////////////////////////////////////////////////////////

#define BUTTON_PIN  A2

#define WIFI_SSID    "Hausbiz2"
#define WIFI_PASSWD  "WeMissMoody"

/////////////////////////////////////////////////////////////////

SimpleFSM fsm;
Button2 btn;

#if defined(ARDUINO_ARCH_ESP32)
  WebServer server(80);
#else
  ESP8266WebServer server(80);
#endif

/////////////////////////////////////////////////////////////////

bool connectToWiFi(char* ssid, char* passwd, int timeout_after =-1) {
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, passwd);
  // Wait for connection
  long now = millis();
  while ((WiFi.status() != WL_CONNECTED) && ((timeout_after > -1) && (millis() < (now + timeout_after)))) {
    delay(500);
  }
  return (WiFi.status() == WL_CONNECTED);
}

/////////////////////////////////////////////////////////////////
void on_yellow() {
  Serial.println("State: Yellow");
}
 
void on_red() {
  Serial.println("State: RED");
}
 
void on_green() {
  Serial.println("State: GREEN");
}

void on_button_press() {
  Serial.println("BUTTON!");
}

/////////////////////////////////////////////////////////////////

State s[] = {
  State("red light",          on_red),
  State("green light",        on_green),
  State("button pressed", on_button_press),
  State("yellow light",       on_yellow)
};

enum triggers {
  button_was_pressed = 1  
};

Transition transitions[] = {
  Transition(&s[0], &s[2], button_was_pressed)
};

TimedTransition timedTransitions[] = {
  TimedTransition(&s[0], &s[3], 6000),
  TimedTransition(&s[3], &s[1], 1000),
  TimedTransition(&s[1], &s[0], 5000),
  TimedTransition(&s[2], &s[1], 2000)
};

int num_transitions = sizeof(transitions) / sizeof(Transition);
int num_timed = sizeof(timedTransitions) / sizeof(TimedTransition);

/////////////////////////////////////////////////////////////////

void button_handler(Button2 btn) {
  fsm.trigger(button_was_pressed);
}

/////////////////////////////////////////////////////////////////

void showGraph() {
  String message = "";
  message += F("<html>\n");
  message += F("<head>\n");
  message += F("<meta http-equiv='refresh' content='1'>\n");
  message += F("<title>GraphVizArt</title>\n");
  message += F("</head>\n");
  message += F("<body>\n");
  message += F("<a href='http://gravizo.com/'><img src='https://g.gravizo.com/svg?");
  message += fsm.getDotDefinition();
  message += F("'/></a>");
  message += F("</body>\n");
  message += F("</html>\n");
  server.send ( 200, F("text/html"), message);
}

/////////////////////////////////////////////////////////////////

void setup() {
    Serial.begin(9600);
    while (!Serial) {
        delay(300);
    }
  Serial.println();
  Serial.println();
  Serial.println("SimpleFSM - Mixed Transition (Traffic light with button for green)\n");
    
  fsm.add(timedTransitions, num_timed);
  fsm.add(transitions, num_transitions);
  
  fsm.setInitialState(&s[0]);

  btn.begin(BUTTON_PIN);
  btn.setTapHandler(button_handler); 

  if (!connectToWiFi(WIFI_SSID, WIFI_PASSWD, 10000)) {
    Serial.println(F("Could not connect to WiFi."));
  } else {
    Serial.print(F("Open http://"));
    Serial.print(WiFi.localIP());
    Serial.print(F(" in your browser.\n"));

    server.on( "/", showGraph);
    server.onNotFound([](){
      server.send(404, F("text/plain"), F("File Not Found\n"));
    });
    server.begin();
  }
  server.handleClient();
}

/////////////////////////////////////////////////////////////////

void loop() {
  fsm.run();
  btn.loop();
  server.handleClient();
}

/////////////////////////////////////////////////////////////////

snowrodeo, 2022-05-28:
I should add that this is the line where it crashes

fsm.add(timedTransitions,` num_timed);

LennartHennigs, 2022-05-28:
Hey,
thanks for pointing this out! It seems to be problem with the ESP32. 
My M5Stack crashes, too.
Will investigate. it seems to be a problem with the `realloc` command.

LennartHennigs, 2022-05-30:
Hey, just uploaded a fix for the bug. Please give it a try.

snowrodeo, 2022-05-31:
Hey Lennart,

Your fix works!  Many thanks!!

-Erik

closed 2022-05-31 by LennartHennigs