LennartHennigs/SimpleFSM #2: Can't compile examples for ESP32 target

fdufnews, 2022-12-15:
Hi,
I tried compiling some examples for the ESP32 target and I always have errors that I don't `understand.
Whichever example I take I have the same errors.
If I change the target for a UNO the compiler only gives warning but no errors.
Find here after the result of the compilation

Regards

```
Using board 'esp32' from platform in folder: /home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4
Using core 'esp32' from platform in folder: /home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4
Detecting libraries used...
/home/frederic/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32 -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_https_ota -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ringbuf -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ethernet -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/expat -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fatfs -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freemodbus -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freertos -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/heap -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/idf_test -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/jsmn -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/json -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/libsodium -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/log -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/lwip -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mdns -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/micro-ecc -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mqtt -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/newlib -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nghttp -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nvs_flash -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/openssl -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protobuf-c -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protocomm -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/pthread -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/sdmmc -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/smartconfig_ack -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/soc -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spi_flash -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spiffs -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcp_transport -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcpip_adapter -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ulp -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/vfs -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wear_levelling -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wifi_provisioning -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wpa_supplicant -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/xtensa-debug-module -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32-camera -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fb_gfx -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32 -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/variants/esp32 /tmp/arduino-sketch-A14585B53C2DEE0025D56E60E30A340B/sketch/MixedTransitions.ino.cpp -o /dev/null
Alternative pour SimpleFSM.h: [SimpleFSM@1.1.0]
ResolveLibrary(SimpleFSM.h)
  -> candidates: [SimpleFSM@1.1.0]
/home/frederic/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32 -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event -I/home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client -I/ho [... cut ...]
/home/frederic/Arduino/libraries/SimpleFSM/src/Transitions.cpp:9:1: error:   when initialized here [-Werror=reorder]
 AbstractTransition::AbstractTransition() :
 ^
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors

Utilisation de la bibliothèque SimpleFSM version 1.1.0 dans le dossier: /home/frederic/PcFred/Arduino/sketchbook_1.6/Arduino/libraries/SimpleFSM 
Utilisation de la bibliothèque Button2 version 2.2.0 dans le dossier: /home/frederic/PcFred/Arduino/sketchbook_1.6/Arduino/libraries/Button2 
exit status 1

Compilation error: exit status 1
```

LennartHennigs, 2022-12-15:
Hey, 
you are using the Arduino IDE, correct?
It looks like you are using an older version of the ESP32 board libs.
This is your compile message:

```Using board 'esp32' from platform in folder: /home/frederic/.arduino15/packages/esp32/hardware/esp32/1.0.4````

My compile message starts like this:

```Using board 'm5stack-core2' from platform in folder: /Users/lennart/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6```

Could you update the ESP32 libs via the board manager and see what happens when you compile then?
Cheers
l.

LennartHennigs, 2022-12-15:
P.S. Followed my own advice, updated to the latest ESP32 version (2.0.5.) and it also compiled fine.

fdufnews, 2022-12-15:
Thank you for your answer.
I am using Arduino IDE 2.0.3 and in the board manager the current proposed ESP32 board libs is 1.0.4.
Your advice is interesting I will look how I can update to more recent board libs.
Thanks a lot.

fdufnews, 2022-12-18:
If this can help another one.
The root cause to the problem was a bit different. Using the same IDE release, the same simpleFSM library release and compiling the same example code, on one of my computer the compilation was ending with no error and with the reported error on the other one.
Comparing the compilation reports of the 2 computers, I finally found that in one of the IDE in the setup the error report was set to _Default_ on the first one and set to _All_ in the second one. Changing All to Default the error is no more reported, the compilation ends as expected and the code can be downloaded and it runs flawessly

LennartHennigs, 2022-12-18:
hey @fdufnews, 
thanks for the update. but then I will try to replicate the error and still try to fix it.
will keep this open for now.

LennartHennigs, 2022-12-18:
Hey @fdufnews,
went through all the compiler warnings and refactored the code a bit.
There should be no more warnings showing up.
Thanks again for pointing them out.

Cheers
l.

fdufnews, 2022-12-18:
You are very reactive that's fine.
I think there is a side effect to your refactoring.
- The SimpleTransitions example works as expected
- The TimedTransitions example makes one transition (from RED to GREEN) and then get stuck in GREEN.
```
SimpleFSM - Timed Transition (Simple traffic light)

State: RED
State: GREEN
```

fdufnews, 2022-12-18:
I have made all the variables public in the library in order to see how the fsm is working
I have changed the loop() in TimedTansitions example to this one
```
unsigned long lastTime = millis();

void loop() {
  
  fsm.run();
  if(millis() - lastTime > 1000){
    lastTime = millis();
    Serial.println(fsm.getState()->getName());
    int numT = fsm.num_timed;
    for(int i = 0; i < numT; i++){
      Serial.print(i); Serial.print(": ");
      Serial.print("start = "); Serial.print(fsm.timed[i].start);
      Serial.print(" interval = "); Serial.print(fsm.timed[i].interval);
      Serial.print(" from: "); Serial.print(fsm.timed[i].from->getName());
      Serial.print(" to: "); Serial.println(fsm.timed[i].to->getName());
    }
  }
}
```

In the console I get this
```
SimpleFSM - Timed Transition (Simple traffic light)

0: start = 0 interval = 6000
1: start = 0 interval = 4000
****************************************
State: RED
red
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
red
0: start = 1046 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
red
0: start = 1046 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
red
0: start = 1046 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
red
0: start = 1046 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
red
0: start = 1046 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
red
0: start = 1046 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
State: GREEN
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
green
0: start = 0 interval = 6000 from: red to: green
1: start = 0 interval = 4000 from: green to: red
```
To my understanding of how run() works, when changing from RED to GREEN
- start of RED shall set to 0
- start of GREEN shall be set to the current value of millis()
But for a reason that I can't actually understand, start of GREEN never changes

fdufnews, 2022-12-18:
I had not seen that you have corrected the bug.
In the mean time I have found that the problem was the break in run() that was preventing to modify any transition but the first one.
I corrected the bug by changing the break to continue.

LennartHennigs, 2022-12-19:
Yes, sorry, I did not ping you to let you know. After you posted your feedback, I knew what the problem was. 

I am in the process of refactoring the code a bit more. 

I will keep this ticket open until I am happy with the changes.

fdufnews, 2022-12-19:
Don't worry. It is not a problem.
As I said it was an interesting quest in your code.

LennartHennigs, 2022-12-19:
:-)

Just released a new version.
Refactored code a but more and found a bug that I fixed.
Put the `continue` back.
Will close this now.
Thanks again!

closed 2022-12-19 by LennartHennigs