LennartHennigs/SimpleFSM #3: Issue with timedTransitions using PlatformIO

gsicilia82, 2023-05-16:
I was looking for a FSM with best user interface, and I really like yours.
After some tests with your examples, by compiling with PlatformIO, I encountered an issue:

(To be clear: everything works well with Arduino-IDE! I don't understand this different behaviour with PlatformIO)

Usually, the code line `fsm.add(timedTransitions, num_timed);` lets crash my esp32:
[image]

Sometimes, the code is running fine until first hardware reset. After first reset, each program start ends up with same crash.

There is one exception about the crash: Adding timedTransitions after from adding standard transitions works well.

This crashs my esp32
```
  fsm.add(timedTransitions, num_timed);
  fsm.add(transitions, num_transitions);
```

This works well
```
  fsm.add(transitions, num_transitions);
  fsm.add(timedTransitions, num_timed);
```

Do you have any idea?

I am working with PlatformIO on different projects, usually it doesn't matter if compiling with PIO or Arduino-IDE. I want to use your FSM in my next project, therefore, I would like to know about this strange behaviour.

Thank you.

LennartHennigs, 2023-05-17:
Hey, thanks for reaching out. And thank you for pointing out that the version is not on platform.io. That's strange. Will check it out. 

And gals that you like the lib. Wasn't also happy with most others. That's why I did mine back then. 😃

closed 2023-05-20 by LennartHennigs

lewsut, 2023-08-03:
still v1.0.0 on platformIO

LennartHennigs, 2023-08-05:
Hey, thanks for pointing it out. 
Just published the new recent version to PlatformIO.