LennartHennigs/SimpleFSM #17: Set TimedTransition interval on setup or loop

danielcharrua, 2024-01-14:
Hello, I'm trying to set the `TimedTransition` interval on `setup()` or `loop()` functions without success.

This is because I load a config.json with some values and some transition interval are defined there, so I need to set the interval on those functions.

Any ideas on how to achieve this?

Thank you.

LennartHennigs, 2024-01-14:
Hey,
Thank you for your feedback. 
Will look into it and get back to you. 

Cheers

danielcharrua, 2024-01-14:
Would be awesome to set this from setup() or loop() functions to add some dynamic timers without using external ones.
Thank you Lennart! 🚀

LennartHennigs, 2024-01-15:
Hey, 
what do you mean with dynamic timers? 
Do you want to add timed transitions in `setup()`?
This does work:

```
TimedTransition timedTransitions[2];

void setup() {
 timedTransitions[0] = TimedTransition(&s[0], &s[1], 6000);
 timedTransitions[1] = TimedTransition(&s[1], &s[0], 4000);
}
```

Or do you want to solve a different problem?

danielcharrua, 2024-01-15:
hey!

I need to read a JSON config file from internal memory with some values including the intervals of the timed transitions on the FSM. I have crafted my logic into your example of TimedTransitions to see the issue I have.

If I load this code the TimedTransitions got 0 as time interval, I think that when the FSM is initialized the values of `turnedOnTimeRed` and `turnedOnTimeGreen` are not set.

My rookie approach thought that setting those variables before `fsm.add(timedTransitions, num_timed);` would do the job and send those intervals but no luck.

```
#include "SimpleFSM.h"

/////////////////////////////////////////////////////////////////

int    turnedOnTimeRed;
int    turnedOnTimeGreen;
SimpleFSM fsm;

/////////////////////////////////////////////////////////////////

void on_red() {
  Serial.println("\nState: RED");
}
 
void on_green() {
  Serial.println("\nState: GREEN");
}

void ongoing() {
  Serial.print(".");
}
/////////////////////////////////////////////////////////////////

State s[] = {
  State("red", on_red /*, ongoing */),
  State("green", on_green, ongoing)
};

TimedTransition timedTransitions[] = {
  TimedTransition(&s[0], &s[1], turnedOnTimeRed),
  TimedTransition(&s[1], &s[0], turnedOnTimeGreen),
};

int num_timed = sizeof(timedTransitions) / sizeof(TimedTransition);

/////////////////////////////////////////////////////////////////

void setup() {
  Serial.begin(9600);
  while (!Serial) {
    delay(300);
  }
  Serial.println();
  Serial.println("SimpleFSM - Timed Transition (Simple traffic light)\n");

  readConfigFromJson();
    
  fsm.add(timedTransitions, num_timed);
  fsm.setInitialState(&s[0]);
}

/////////////////////////////////////////////////////////////////

void loop() {
  fsm.run();
}

/////////////////////////////////////////////////////////////////

void readConfigFromJson()
{
  // read and set variables...
  turnedOnTimeRed = 6000;
  turnedOnTimeGreen = 4000;
}

```

danielcharrua, 2024-02-08:
@LennartHennigs any updates on this issue? Thank you.