LennartHennigs/SimpleFSM #22: question on the use of triggers

ator1811, 2024-12-05:
Hi, I have a question with regards to the use of triggers. Best I describe it with an example:

I have 3 states, called 1, 2 and 3. I defined a trigger to go from state 2 to state 3. For some reason the trigger is triggered in state 1. What happens when the state machine comes in state 2? Will it directly go to state 3 (because a trigger is present), or does it wait until the trigger is triggered (neglating the previous trigger)?

I hope it is clear what I mean.
Thanks

LennartHennigs, 2024-12-14:
Hey. 
That is the beauty of state machines. No conditions (if/else) are needed for a state machine to determine where to transition to. It is defined by the states. With the wrong trigger nothing happens.

closed 2024-12-14 by LennartHennigs

ator1811, 2024-12-15:
Thanks for your response. Sorry to bother again, but I did some testing and I get results I don't understand.

I used the example script MixedTransistion.ino. I added a Serial print line in the button_handler function to indicate a button press
```
  void button_handler(Button2 &btn) {
  Serial.print("button pressed"); //<-- added
  fsm.trigger(button_was_pressed);
}
```
I pressed the button during the Red, the Green state(2x). Below the output of the script. I added some remarks to it. "button pressed" is the actual button press.

```
....
State: GREEN
.....
State: RED
.......
State: GREEN
.....
State: RED
....button pressed
State BUTTON!  <== as expected. In state Red it jumps to state Button
.
State: GREEN
....button pressed.  <== button press during the green state
State: RED
. <==  state Red jumps to state Green directly
State: GREEN
.....
State: RED
.......
State: GREEN
..button pressed...<== again button press during the green state
State: RED
....... <== now it keeps in the Red state
State: GREEN
.....
State: RED
.....

```
A button press during the green state is handled different in the red state. Is this how it should work or is there a bug? Thanks for your response.

ator1811, 2024-12-16:
Thanks for your response. Sorry to bother again, but I did some testing and I get results I don't understand.

I used the example script MixedTransistion.ino. I added a Serial print line in the button_handler function to indicate a button press and the trigger initiation
```
  void button_handler(Button2 &btn) {
  Serial.print("button pressed"); //<-- added
  fsm.trigger(button_was_pressed);
}
```
I pressed the button during the Red, the Green state(2x). Below the output of the script. I added some remarks to it. "button pressed" is the actual button press.
```
....
State: GREEN
.....
State: RED
.......
State: GREEN
.....
State: RED
....button pressed
State BUTTON!  <== as expected. In state Red it jumps to state Button
.
State: GREEN
....button pressed.  <== button press during the green state. Nothing happened as expected
State: RED
. <==  state Red jumps to state Green directly without waiting for the 6000 ms
State: GREEN
.....
State: RED
.......
State: GREEN
..button pressed...<== again button press during the green state, nothing happened as expected
State: RED
....... <== now it keeps in the Red state for the full 6000 ms
State: GREEN
.....
State: RED
.....
```
A button press during the green state is handled differently in the red state, based on the history. Is this how it should work or is there a bug? Thanks for your response.

LennartHennigs, 2024-12-21:
Please look at the .svg of the example. I think it explains well how it works. 
Also, your println() might take too long so that timings are off.