LennartHennigs/SimpleFSM #27: fsm.getState()->getName() sometimes returns garbage

pmi123, 2025-07-04:
I am working on an ESP32-S3 project using SimpleFSM. I have run into an issue with printing the name of the current state. In all cases after state 3, the name is garbage characters but the state ID is correct. I shrunk my actual project from ~4,000 lines of code to 336 lines in one ino file, and the problem still exists. I have callbacks (`on_enter`, `on_state`, and `on_exit`) for each state. In each `on_state` I have this code:

```
  if (!messagePrinted) {
    State* s = fsm.getState();
    Serial.printf("Living in %s, ID=%d\n", s->getName(), s->getID());
    messagePrinted = true;
  }
```
`messagePrinted = false` at the top of the file, and `messagePrinted` is set back to `false` in the `on_exit` callback.

Here is a sample output. Note that State 0 prints the name and ID correctly, as well as state 1, 2, and 3. But states 4 and 5 return garbage for the state name. 
```
Type 0, then return to get to the initial state. Then type state number to go to that state. You must start at state 0 each time.
Entering START
Living in START, ID=0
State 1
Leaving START
Entering LAUNCH
Living in LAUNCH, ID=1
State 0
Leaving LAUNCH
Entering START
Living in START, ID=0
State 2
Leaving START
Entering SETUP
Living in SETUP, ID=2
State 0
Leaving SETUP
Entering START
Living in START, ID=0
State 3
Leaving START
Entering HELP
Living in HELP, ID=3
State 0
Leaving HELP
Entering START
Living in START, ID=0
State 4
Leaving START
Entering LAUNCH
Leaving LAUNCH
Entering LAUNCHER READY
Living in P��?, ID=4
State 0
Leaving LAUNCHER READY
Entering START
Living in START, ID=0
State 5
Leaving START
Entering LAUNCH
Leaving LAUNCH
Entering LAUNCHER ERROR
Living in P��?, ID=5
State 0
Leaving LAUNCHER ERROR
Entering START
Living in START, ID=0
State 1
Leaving START
Entering LAUNCH
Living in LAUNCH, ID=1
State 0
Leaving LAUNCH
Entering START
Living in START, ID=0
State 2
Leaving START
Entering SETUP
Living in SETUP, ID=2
```
I can get around the problem by not calling `fsm.getState()`, but it would be nice if I could!

LennartHennigs, 2025-07-05:
Hey, thanks for reporting. What is the baud speed you’re using?Could you try 115200?It might be that it takes too long to output the name.

pmi123, 2025-07-05:
I am using 115200. It is in the .ino file I sent you, line 251. Let me know if I can help in any other way.